queezythegreat/arduino-cmake #108: RF24 Arduino library fails to build

thirtythreeforty, 2015-01-22:
I have run into an issue when trying to build the [RF24](https://github.com/TMRh20/RF24) Arduino library.  According to some other issues I've read, Arduino library dependencies are supposed to be detected automatically, but as far as I can tell that isn't happening correctly.

The output looks like this:

```
[georgev@shillig-arch firmware (master)]$ mdcd build
[georgev@shillig-arch build (master)]$ cmake ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Arduino SDK version 1.0.6: /usr/share/arduino
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works

(snipped)

[ 88%] Building CXX object src/CMakeFiles/nano328_src.dir/blink.cpp.obj
[ 92%] Building CXX object src/CMakeFiles/nano328_src.dir/blink_lib.cpp.obj
Linking CXX static library libnano328_src.a
[ 92%] Built target nano328_src
Scanning dependencies of target blink_lib
[ 96%] Building CXX object src/CMakeFiles/blink_lib.dir/blink_lib.cpp.obj
Linking CXX static library libblink_lib.a
[ 96%] Built target blink_lib
Scanning dependencies of target blink
[100%] Building CXX object src/CMakeFiles/blink.dir/blink.cpp.obj
In file included from /home/georgev/classes/Senior Design/firmware/libraries/RF24/RF24.h:18:0,
                 from /home/georgev/classes/Senior Design/firmware/src/blink.cpp:2:
/home/georgev/classes/Senior Design/firmware/libraries/RF24/RF24_config.h:104:19: fatal error: SPI.h: No such file or directory
   #include <SPI.h>
                   ^
compilation terminated.
src/CMakeFiles/blink.dir/build.make:54: recipe for target 'src/CMakeFiles/blink.dir/blink.cpp.obj' failed
make[2]: *** [src/CMakeFiles/blink.dir/blink.cpp.obj] Error 1
CMakeFiles/Makefile2:92: recipe for target 'src/CMakeFiles/blink.dir/all' failed
make[1]: *** [src/CMakeFiles/blink.dir/all] Error 2
Makefile:76: recipe for target 'all' failed
make: *** [all] Error 2
[georgev@shillig-arch build (master)]$ 
```

[My sample repository](https://github.com/thirtythreeforty/arduino-cmake-issue) demonstrates the issue.  You will have to run a recursive clone if you want a copy, as I'm pulling the dependencies (including this repository itself) in via submodules.

Am I doing something incorrectly, or is there an issue?

aditanase, 2015-02-10:
Have you tried using link_directories to make sure the dependencies are loaded from the correct location? 
`link_directories(/Users/$ENV{USER}/Documents/Arduino/libraries/)`

Also, if a library has dependencies on standard libs, you may need to configure recursive scan of dependencies. In some of my sketches I needed this for the aJSON library. In your case, it might be:
`set(RF24_RECURSE True)`

Both directives need to be placed after the project definition, before the `generate_arduino_firmware` call that needs the libs in question.

thirtythreeforty, 2015-07-21:
I haven't messed with this in a while, but I seem to recall that I was doing something incorrectly.  I'll close this if you don't mind.

closed 2015-07-21 by thirtythreeforty