queezythegreat/arduino-cmake #146: Doesn't find any board configuration

narrowtux, 2015-04-10:
I have a very simple project configuration, posted on this [Gist](https://gist.github.com/narrowtux/1bf0d05276356e245984).

As you can see, I reconfigured the example config a bit but everything else is still at the same place (toolchain and `Arduino.cmake` files in `cmake/...`).

When I execute cmake from the `build/` directory with `cmake ..`, it will throw this error:

```
Scarecrow:build tux$ cmake ..
--
-- Generating testtesttest
CMake Error at cmake/Platform/Arduino.cmake:830 (message):
  Invalid Arduino board ID (uno), aborting.
Call Stack (most recent call first):
  cmake/Platform/Arduino.cmake:1071 (get_arduino_flags)
  cmake/Platform/Arduino.cmake:516 (setup_arduino_target)
  testtesttest/CMakeLists.txt:8 (generate_arduino_firmware)

-- Configuring incomplete, errors occurred!
See also "/Users/tux/Downloads/arduino-cmake-master/build/CMakeFiles/CMakeOutput.log".
See also "/Users/tux/Downloads/arduino-cmake-master/build/CMakeFiles/CMakeError.log".
```

As you can see, it can't find the configuration for the Arduino Uno. In fact, it doesn't have any configuration for any other board (since you can see `print_board_list()` doesn't list anything.

I dug a bit with `cmake --trace ..` and saw that `register_hardware_platform(${ARDUINO_SDK_PATH}/hardware/arduino/)` is never executed.

Can I change something in my configuration so it will load the `boards.txt` file?

hanikesn, 2015-04-10:
This happens because the master branch is currently not compatible with the Arduino SDK 1.5/1.6.

I had more luck with:
https://github.com/blemasle/arduino-cmake
But you need to set ARDUINO_1_5 to True.

Honestly, Ardunio development is a huge mess right now ... It all needs to settle down a bit.

narrowtux, 2015-04-10:
Thanks, that worked :)

murray-lang, 2017-02-01:
For the sake of people still having a problem with this, I have found that changing
set(${PROJECT_NAME}_BOARD pro) 
to
set(${PROJECT_NAME}_BOARD uno)
or any other board fails unless I comment out
set(ARDUINO_CPU 16MHzatmega328).
It appears that setting the CPU causes problems if the board doesn't support multiple CPUs

axelbr, 2018-09-07:
I think the issue is caused by the folder structure of the arduino sdk. The boards.txt is located in `your_path_to_sdk/hardware/arduino/avr`. However, in the CMakeLists.txt, the path where the boards.txt file is required to be is: `your_path_to_sdk/hardware/arduino`. 
Changing this worked for me. I am using arduino sdk 1.8.6.

maartendemunck, 2018-12-13:
The issue mentioned by @axelbr can be solved by adding
`register_hardware_platform(${ARDUINO_SDK_PATH}/hardware/arduino/avr)`
to your CMakeLists.txt (so you shouldn't even patch arduino-cmake itself).
I know it's an old thread, but I found this page by googling the error message, so I wanted to share this solution for other people landing here.

nnarain, 2019-06-13:
Thank you @maartendemunck. That worked for me.

schlegel11, 2020-05-21:
I came across this issue looking for a solution with the Clion arduino plugin on arch linux.
The solution here was to create a soft link for the arduino hardware platform path which is usually 
`/usr/share/arduino/hardware/archlinux-arduino/avr` and should be `/usr/share/arduino/hardware/arduino/avr`.
It was also necessary to recreate the Clion arduino project. With an existing project which I imported from my old linux environment it didn't worked due to ` register_hardware_platform(${ARDUINO_SDK_PATH}/hardware/arduino/avr)` was never called.
Im currently not sure why it works with a new created Clion arduino project but anyway it works :smile:

gsilvan, 2020-05-30:
Thanks @schlegel11. I had to recreate the project as well.