queezythegreat/arduino-cmake #157: Invalid Arduino board ID (mega2560), aborting.

ronny332, 2015-10-17:
Hi,
until today I was using arduino-cmake together with a plugin for the CLion IDE by JetBrains together with and Uno Board (328p). My OS is OS X 10.10 and 10.11.

Today a mega2560 based board arrived and I was trying to use it with arduino-cmake (the Arduino IDE detects and uses it with issues).

The compilation is not working to the end, it seems like the mega2560 setup is not working:

```
Invalid Arduino board ID (mega2560), aborting.
```

Meanwhile I've tried to use arduino-cmake without the CLion Plugin (latest version + lastest sdk), the error is exactly the same.

Is this a known issue or is the mega2560 chip simply not supported? It exists as example even at the main documentation.

rpetz, 2015-10-26:
You'll need to show your basic setup in your cmakelists.txt file before we could figure out why it's not working for you.  I've been working through similar issues with the ESP8266 board so I might be able to help if you're still having issues.

ronny332, 2015-10-26:
@rpetz,

meanwhile I got it working. In my case I use arduino-cmake together with the CLion Plugin of Arduino.

With these settings the 2560 is running very well.

```
cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME avr_test)
project(${PROJECT_NAME})
set(${CMAKE_PROJECT_NAME}_BOARD mega)
set(${CMAKE_PROJECT_NAME}_PORT /dev/tty.usbmodem142141)
set(${CMAKE_PROJECT_NAME}_SKETCH avr_test.ino)
set(mega.build.mcu atmega2560)
set(mega.upload.protocol wiring)
set(mega.upload.speed 115200)
generate_arduino_firmware(${CMAKE_PROJECT_NAME})
```

Fedtekansler, 2016-04-07:
I am getting the same error, you had. I have tried running it with the settings you specified as well.

Did you change anything else to make it work? My configuration looks like this:

`cmake_minimum_required(VERSION 2.8.4)
set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME test)
project(${PROJECT_NAME})
set(${CMAKE_PROJECT_NAME}_BOARD mega2560)
set(${CMAKE_PROJECT_NAME}_PORT /dev/cu.usbmodem1421)
set(${CMAKE_PROJECT_NAME}_SKETCH test.ino)
generate_arduino_firmware(${CMAKE_PROJECT_NAME})`

kerbymart, 2016-11-28:
@ronny332 are these lines required?

set(mega.upload.protocol wiring)
set(mega.upload.speed 115200)

?

Grandmother, 2018-01-05:
As mentioned [here](https://github.com/queezythegreat/arduino-cmake/issues/180) project is still alive and not it lives [here](https://github.com/arduino-cmake/arduino-cmake).

Here cmake probably can't find a path to boards.txt file.

Grandmother, 2018-01-05:
I've solved my problem on archlinux by creating a symbolic link to ```/usr/share/arduino/hardware/archlinux-arduino/```:
```
# ln -s /usr/share/arduino/hardware/archlinux-arduino/ /usr/share/arduino/hardware/arduino
```

razorback1, 2018-07-31:
Was following http://wiki.ros.org/rosserial_arduino/Tutorials/CMake and had this problem, changed firmware/CMakeLists.txt to 'BOARD mega2560' instead of 'BOARD leonardo'. avrdude seems happy now. Continuing on my learning curve.

ProtonPunk, 2019-02-22:
I've had a similar issue on debian(9.8) where any board architecture could not be found due to using both tool-chains at once. The following solved the issue:
- deinstalling the system tools: `gcc-avr`,`binutils-avr`, `avr-libc`, `avrdude`. Avr-tools are bundled with the ArduinoSDK under `arduino-1.8.X/hardware/tools/avr/bin`
- Ignoring ARDUINO_SDK_PATH and installing SDK under `/usr/share/arduino`.