queezythegreat/arduino-cmake #158: Problems with linking when using U8glib library (works without problems on Arduino IDE)

dawidm, 2015-10-18:
I have Arduino IDE 1.0.5, cmake 3.0.2.
I downloaded library https://bintray.com/olikraus/u8glib/Arduino and put it on /usr/share/arduino/libraries. It works well with arduino ide, but doesn't link with arduino cmake. Example (minimal) code:
```
#include <U8glib.h>
U8GLIB_SH1106_128X64 u8g(13, 11, 10, 9, 8);	// SW SPI Com: SCK, MOSI, CS, A0
void setup() {
}
void loop() {
}
```
Errors i get with arduino-cmake:
```
Linking CXX static library libuno_CORE.a
[ 86%] Built target uno_CORE
Scanning dependencies of target uno_U8glib
[ 91%] Building CXX object CMakeFiles/uno_U8glib.dir/usr/share/arduino/libraries/U8glib/U8glib.cpp.obj
Linking CXX static library libuno_U8glib.a
[ 91%] Built target uno_U8glib
Scanning dependencies of target untitled
[ 95%] Building CXX object CMakeFiles/untitled.dir/untitled_untitled.ino.cpp.obj
Linking CXX executable untitled.elf
CMakeFiles/untitled.dir/untitled_untitled.ino.cpp.obj: In function `_GLOBAL__sub_I_u8g':
/usr/share/arduino/hardware/arduino/cores/arduino/Print.h:43: undefined reference to `u8g_DrawGlyph'
CMakeFiles/untitled.dir/untitled_untitled.ino.cpp.obj: In function `_GLOBAL__sub_I_u8g':
/home/dawid/ClionProjects/untitled/untitled.ino:129: undefined reference to `u8g_dev_sh1106_128x64_sw_spi'
/home/dawid/ClionProjects/untitled/untitled.ino:129: undefined reference to `u8g_dev_sh1106_128x64_sw_spi'
libuno_U8glib.a(U8glib.cpp.obj): In function `U8GLIB::initRW8Bit(_u8g_dev_t*, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char)':
/usr/share/arduino/libraries/U8glib/U8glib.cpp:75: undefined reference to `u8g_InitSPI'
collect2: error: ld returned 1 exit status
CMakeFiles/untitled.dir/build.make:92: recipe for target 'untitled.elf' failed
make[2]: *** [untitled.elf] Error 1
CMakeFiles/Makefile2:131: recipe for target 'CMakeFiles/untitled.dir/all' failed
make[1]: *** [CMakeFiles/untitled.dir/all] Error 2
Makefile:76: recipe for target 'all' failed
make: *** [all] Error 2
```
CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.0)

set(CMAKE_TOOLCHAIN_FILE ${CMAKE_SOURCE_DIR}/cmake/ArduinoToolchain.cmake)
set(PROJECT_NAME untitled)
project(${PROJECT_NAME})

set(${CMAKE_PROJECT_NAME}_SKETCH untitled.ino)
generate_arduino_firmware(${CMAKE_PROJECT_NAME}
BOARD uno
PORT /dev/ttyUSB0
SERIAL picocom @SERIAL_PORT@ -b 9600 -l)
```

ad-on-is, 2015-10-31:
having exactly the same problem
````
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::write(unsigned char)':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:94: undefined reference to `u8g_DrawGlyph'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::setFont(unsigned char const*)':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:169: undefined reference to `u8g_font_unifont'
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:169: undefined reference to `u8g_font_unifont'
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:169: undefined reference to `u8g_SetFont'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::drawStr(unsigned char, unsigned char, char const*)':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:174: undefined reference to `u8g_DrawStr'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::setColorIndex(unsigned char)':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:124: undefined reference to `u8g_SetColorIndex'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::setHiColorByRGB(unsigned char, unsigned char, unsigned char)':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:121: undefined reference to `u8g_SetHiColorByRGB'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::cbegin()':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:53: undefined reference to `u8g_Begin'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::firstPage()':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:110: undefined reference to `u8g_FirstPage'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB::nextPage()':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:111: undefined reference to `u8g_NextPage'
CMakeFiles/cltest.dir/cltest_cltest.ino.cpp.obj: In function `U8GLIB':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:78: undefined reference to `u8g_dev_ssd1306_128x64_i2c'
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.h:78: undefined reference to `u8g_dev_ssd1306_128x64_i2c'
libuno_U8glib.a(U8glib.cpp.obj): In function `U8GLIB::initI2C(_u8g_dev_t*, unsigned char)':
/Volumes/adONis/Documents/Projekte/Arduino/libraries/U8glib/U8glib.cpp:57: undefined reference to `u8g_InitI2C'
collect2: error: ld returned 1 exit status
make[3]: *** [cltest.elf] Error 1
make[2]: *** [CMakeFiles/cltest.dir/all] Error 2
make[1]: *** [CMakeFiles/upload.dir/rule] Error 2
make: *** [upload] Error 2
````

ad-on-is, 2015-11-11:
The solution is described here: https://github.com/queezythegreat/arduino-cmake#creating-libraries
Simply add this to your CMakeLists.txt
````
link_directories(/your/path/to/U8glib)
set(U8glib_RECURSE True)
````

PNixx, 2017-11-22:
I try error on Arduino 1.6.0 and CLion 2017.2.3:

```
/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build ./cmake-build-debug --target Camera_v2 -- -j 2
[ 89%] Built target nano_CORE
Scanning dependencies of target Camera_v2
[ 92%] Building CXX object CMakeFiles/Camera_v2.dir/Camera_v2_Camera_v2.ino.cpp.obj
[ 96%] Linking CXX executable Camera_v2.elf
CMakeFiles/Camera_v2.dir/Camera_v2_Camera_v2.ino.cpp.obj: In function `setup':
./Camera_v2/Camera_v2.ino:51: undefined reference to `A4988::begin(short, short)'
./Camera_v2/Camera_v2.ino:52: undefined reference to `BasicStepperDriver::enable()'
CMakeFiles/Camera_v2.dir/Camera_v2_Camera_v2.ino.cpp.obj: In function `__static_initialization_and_destruction_0':
./Camera_v2/Camera_v2.ino:31: undefined reference to `DRV8825::DRV8825(short, short, short, short, short, short, short)'
collect2: error: ld returned 1 exit status
gmake[3]: *** [CMakeFiles/Camera_v2.dir/build.make:100: Camera_v2.elf] Error 1
gmake[2]: *** [CMakeFiles/Makefile2:164: CMakeFiles/Camera_v2.dir/all] Error 2
gmake[1]: *** [CMakeFiles/Makefile2:176: CMakeFiles/Camera_v2.dir/rule] Error 2
gmake: *** [Makefile:157: Camera_v2] Error 2
```