queezythegreat/arduino-cmake #160: Reset before Upload

peterquiel, 2015-12-10:
The arduino micro rests if an serial connection is open with 1200 baud an closed again. 
A software reset before uploading makes uploading a new firmware a one click/command action.

A command like 
```
stty /dev/device 1200 && sleep 3 
```
before uploading did the job for me.

Would be great to have an optional reset command parameter that is executed before uploading the firmeware.

m42e, 2016-01-07:
```diff
diff --git a/cmake/Platform/Arduino.cmake b/cmake/Platform/Arduino.cmake
index e5003e1..e0c412b 100644
--- a/cmake/Platform/Arduino.cmake
+++ b/cmake/Platform/Arduino.cmake
@@ -1208,10 +1208,14 @@ function(setup_arduino_bootloader_upload TARGET_NAME BOARD_ID PORT AVRDUDE_FLAGS

     list(APPEND AVRDUDE_ARGS "-Uflash:w:${TARGET_PATH}.hex")
     list(APPEND AVRDUDE_ARGS "-Ueeprom:w:${TARGET_PATH}.eep:i")
+    set (RESET_COMMAND stty -f ${PORT} 1200 && sleep 2 )
+    add_custom_target(${UPLOAD_TARGET}-reset
+                        ${RESET_COMMAND}
+        )
     add_custom_target(${UPLOAD_TARGET}
                      ${ARDUINO_AVRDUDE_PROGRAM}
                      ${AVRDUDE_ARGS}
-                     DEPENDS ${TARGET_NAME})
+                     DEPENDS ${TARGET_NAME} ${UPLOAD_TARGET}-reset)

     # Global upload target
     if(NOT TARGET upload)
```

oldmud0, 2017-04-07:
The only problems is that `stty` doesn't work on Windows.