queezythegreat/arduino-cmake #168: External SDK path does not work

thomasgt, 2017-01-31:
Manually setting the Arduino SDK path does not work. This issue can only replicated in environments where the Arduino IDE is not installed in one of the default locations.

I am working in Ubuntu 16.04. The only Arduino IDE/SDK on my system is installed in /opt/tools/arduino-1.0.3. I am running cmake from an out-of-tree build directory like this: 
`cmake -DARDUINO_SDK_PATH=/opt/tools/arduino-1.0.3 ..`

I see the following output + error messages:
```
-- The C compiler identification is GNU 4.3.2
-- The CXX compiler identification is GNU 4.3.2
-- Arduino SDK version 1.0.3: /opt/tools/arduino-1.0.3
-- Check for working C compiler: /opt/tools/arduino-1.0.3/hardware/tools/avr/bin/avr-gcc
CMake Error at /home/thomas/Documents/arduino-cmake-master/cmake/ArduinoToolchain.cmake:81 (message):
  Could not find Arduino SDK (set ARDUINO_SDK_PATH)!
Call Stack (most recent call first):
  /home/thomas/Documents/arduino-cmake-master/build/CMakeFiles/3.5.1/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:3 (project)

CMake Error: Internal CMake error, TryCompile configure of cmake failed
-- Configuring incomplete, errors occurred!
See also "/home/thomas/Documents/arduino-cmake-master/build/CMakeFiles/CMakeOutput.log".
```

If I install the arduino IDE in a different location, such as /opt/arduino-1.0.3, it works, but only because this is one of the default locations. It effectively ignores the path I specified when I invoked cmake.

It doesn't fail immediately (it finds the compiler), which is strange. It appears that the ARDUINO_SDK_PATH variable is being cleared at some point, but I can't figure out where this is happening.

thomasgt, 2017-01-31:
I think the issue is related to two conditions in cmake/ArduinoToolchain.cmake. The DEFINED keyword should be use to detect the ARDUINO_SDK_PATH:
`if(NOT DEFINED ARDUINO_SDK_PATH)`
and 
`if(DEFINED ARDUINO_SDK_PATH)`

frans-fuerst, 2017-05-01:
I've stumbled over this, too, but I think is has to do with compiler check. Looks like the compiler check step just uses it's own variable space. 

(also see my StackOverflow question I've asked right now: http://stackoverflow.com/questions/43718042/how-can-a-cmake-variable-be-hidden)

MrPointer, 2017-12-14:
If still relevant, please see #180. This should resolve all those issues.