queezythegreat/arduino-cmake #175: Erro avr-g++: error: missing device or architecture after ‘-mmcu=’

rafaelgov95, 2017-05-19:
# **_When running upload, it gives the following error_**

```
/opt/clion-2017.1.1/bin/cmake/bin/cmake --build /home/rafael/arduino-cmake/cmake-build-debug --target upload -- -j 4
[  7%] Building CXX object example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/CDC.cpp.obj
[  7%] Building CXX object example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HID.cpp.obj
[ 14%] Building CXX object example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj
[ 14%] Building CXX object example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj
avr-g++: error: missing device or architecture after ‘-mmcu=’
avr-g++: error: missing device or architecture after ‘-mmcu=’
avr-g++: error: missing device or architecture after ‘-mmcu=’
example/CMakeFiles/mega_CORE.dir/build.make:86: recipe for target 'example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HID.cpp.obj' failed
make[3]: *** [example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HID.cpp.obj] Error 1
make[3]: ** Esperando que outros processos terminem.
example/CMakeFiles/mega_CORE.dir/build.make:62: recipe for target 'example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/CDC.cpp.obj' failed
make[3]: *** [example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/CDC.cpp.obj] Error 1
example/CMakeFiles/mega_CORE.dir/build.make:110: recipe for target 'example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj' failed
make[3]: *** [example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HardwareSerial.cpp.obj] Error 1
avr-g++: error: missing device or architecture after ‘-mmcu=’
example/CMakeFiles/mega_CORE.dir/build.make:134: recipe for target 'example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj' failed
make[3]: *** [example/CMakeFiles/mega_CORE.dir/usr/share/arduino-1.6.3/hardware/arduino/avr/cores/arduino/HardwareSerial0.cpp.obj] Error 1
CMakeFiles/Makefile2:223: recipe for target 'example/CMakeFiles/mega_CORE.dir/all' failed
make[2]: *** [example/CMakeFiles/mega_CORE.dir/all] Error 2
CMakeFiles/Makefile2:129: recipe for target 'example/CMakeFiles/upload.dir/rule' failed
make[1]: *** [example/CMakeFiles/upload.dir/rule] Error 2
Makefile:131: recipe for target 'upload' failed
make: *** [upload] Error 2

```

**/exemple/CMakeLists.txt**
```

# Set a variable for commands below
set(PROJECT_NAME example)
set(ARDUINO_C_FLAGS      "-ffunction-sections -fdata-sections")
set(ARDUINO_CXX_FLAGS    "${ARDUINO_C_FLAGS} -fno-exceptions")
set(ARDUINO_LINKER_FLAGS "-Wl,--gc-sections")
# Define your project and language
project(${PROJECT_NAME} C CXX)

set(${PROJECT_NAME}_BOARD mega)

# Define the source code
set(${PROJECT_NAME}_SRCS example.cpp)

# Define the port for uploading code to the Arduino
set(${PROJECT_NAME}_PORT /dev/ttyACM0)

# Command to generate code arduino firmware (.hex file)
generate_arduino_firmware(${PROJECT_NAME})

```

**/exemple/exemple.cpp**

```
#include <HardwareSerial.h>
#include "Arduino.h"

#define LED_PIN 13

void setup() {
    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, LOW);

    Serial.begin(9600);
}

void loop() {
    Serial.println("Example arduino project in CLion");

    for (int i = 0; i < 2; i++) {
        digitalWrite(LED_PIN, HIGH);
        delay(100);
        digitalWrite(LED_PIN, LOW);
        delay(100);
    }

    delay(1000);
}
```
## n cpp the Serial function turns red without a library

# I need grateful help!

MrPointer, 2017-05-30:
It is because the 'Mega' board has multiple architectures it can operate on.

As such, you should manually specify the architecture being used by adding the following line to your CMakeLists.txt file:
`set(mega.build.mcu atmega328)`

Note that some other boards such as the 'Nano' also have this "limitation", and therefore should be manually defined.  However, every board has a different ID, and possibly a different architecture, thus have to be defined differently, according to the following pattern:
`set(BOARD_ID.build.mcu ARCHITECTURE)`

rafaelgov95, 2017-05-30:
Put the indicated line but did not do it, continue with the error.

MrPointer, 2017-05-30:
Try to reset CMake's cache instead of just reloading it, should solve the issue.

rafaelgov95, 2017-05-30:
Do you have any design you can share? I think it's a problem in the project!

MrPointer, 2017-05-31:
OK, here is a very basic CMakeLists.txt file that I've to test again the *Nano* board, which also has multiple cpu architectures.

### CMake Script

```CMake
cmake_minimum_required(VERSION 3.7)

set(CMAKE_TOOLCHAIN_FILE cmake/ArduinoToolchain.cmake)
set(ARDUINO_1_5 True)
set(CMAKE_CXX_STANDARD 11)

project(Blink)

set(SOURCE_FILES blink.cpp)

set(${CMAKE_PROJECT_NAME}_SRCS ${SOURCE_FILES})
set(${CMAKE_PROJECT_NAME}_BOARD nano)
# Explicity define board's properties
set(nano.build.mcu atmega328p)
set(nano.upload.speed 57600)

GENERATE_ARDUINO_FIRMWARE(${CMAKE_PROJECT_NAME})
```

### Key-Points

1.  The line `set(ARDUINO_1_5 True)` should be defined **only if** your used SDK version is bigger than 1.5 (Today's most recent version found in the official site is 1.8.2)
2. The line `set(nano.build.mcu atmega328p)` is the one you need. Remember to change *nano* to *mega* in your configuration. Also note that the *p* letter after the processor architecture is not mandatory, and can be omitted.
3. The line `set(nano.upload.speed 57600)` is defined to explicitly set *avrdude*'s uploading speed. The value is taken from the `boards.txt` file located in the *Arduino SDK*, and is currently required to upload something to the connected board. Do note that each board has its own unique baud rate (value).

MrPointer, 2017-12-14:
## Update
An official resurrection of the project exist, including a complete fix to this issue!
Please see #180 :)