queezythegreat/arduino-cmake #177: Arduino SDK v1.8.1 cmake error

jnbrq, 2017-08-02:
I am trying to build provided CMakeLists file. I have an error concerning Wire example doesn't exist. That's fine. But I have a more serious problem (after commenting out wire example target):

```
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Arduino SDK version 1.8.1: /home/janberq/arduino-sdk
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/avr-g++
-- Check for working CXX compiler: /usr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- 
-- Generating blink_original
CMake Error at cmake/Platform/Arduino.cmake:830 (message):
  Invalid Arduino board ID (uno), aborting.
Call Stack (most recent call first):
  cmake/Platform/Arduino.cmake:1071 (get_arduino_flags)
  cmake/Platform/Arduino.cmake:516 (setup_arduino_target)
  example/CMakeLists.txt:35 (generate_arduino_firmware)

-- Configuring incomplete, errors occurred!
See also "/home/janberq/src/arduino-cmake/build/CMakeFiles/CMakeOutput.log".
```
My host system is Ubuntu 17.04 x64. I used following command to compile:

`cmake .. -DARDUINO_SDK_PATH="/home/janberq/arduino-sdk"`

dgrat, 2017-09-15:
@jnbrq 
I get this, and if I comment the lib-wire out I get your error.

```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Arduino SDK version 1.8.3: /usr/share/arduino
-- Check for working C compiler: /usr/bin/avr-gcc
-- Check for working C compiler: /usr/bin/avr-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/avr-g++
-- Check for working CXX compiler: /usr/bin/avr-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- 
-- Generating wire_example
CMake Error at cmake/Platform/Arduino.cmake:1689 (message):
  Could not find example master_writer from library Wire
Call Stack (most recent call first):
  cmake/Platform/Arduino.cmake:614 (setup_arduino_example)
  example/CMakeLists.txt:12 (generate_arduino_example)

-- Configuring incomplete, errors occurred!
See also "/home/dgrat/projects/arduino-cmake/build/CMakeFiles/CMakeOutput.log".
```

MrPointer, 2017-12-14:
Please review #180 as it should address your issues!

duyleomessi, 2018-03-05:
I also got this problem. Has anybody resolved it???