queezythegreat/arduino-cmake #180: The project is alive again!

MrPointer, 2017-12-14:
## Introduction
Hello everyone, I'm one of the maintainers of the next-generation [Arduino CMake](https://github.com/arduino-cmake/arduino-cmake) project.
As years have passed, **this** project has *died*, with no one left to maintain it. 
It is so abandoned that there isn't even a chance to switch maintainers, since the project owner has disappeared. It's a shame that such a great idea (and implementation) would be demised because of this.

However, the community is alive, more then ever, with a lot of great minds contributing to the open source. The [Arduino CMake Organization](https://github.com/arduino-cmake) is a living proof of that.

## Project's Resurrection
We at the **Arduino CMake Organization** have managed to release a stable, fine-grained, up-to-date version of the [Arduino CMake Toolchain](https://github.com/arduino-cmake/arduino-cmake/releases/tag/v2.0.0) by resurrecting the community, adding must-have features and fixing critical bugs that haven't been solved by this project as the original [Arduino Project](https://github.com/arduino/Arduino) has progressed a lot during those years.
We aim to constantly get better, providing support for all kinds of issues, including:
* Support for non-AVR boards
* Support for USB-bound boards
* Performance (Build time & size, CMake script efficiency, etc.)
* And a lot more

## Why Does It Matter?
If you still wonder why I've posted this issue, it is to invite you to join our community and contribute to the **Arduino-CMake** project, which is far from *dead*.
Unfortunately, many people still refer to this repository as the main project to address the **Arduino-CMake** integration, and don't even know that a newer, active development exists. This post attempts and "lure" them, including you, the reader, to use our repository instead. **It is for your best experience, trust me.**

## Summary
To sum it up, the [Arduino CMake Organization](https://github.com/arduino-cmake) invites you to join us in a combined effort to resurrect the **Arduino-CMake** community, and continue the development of the project. We hope to see you soon! 
Made with Love.

queezythegreat, 2017-12-14:
Awesome!

I'm glad to see somebody took interest in my project and is trying to keep it alive.

I unfortunately have had absolutely no time over the past years to maintain this project, and haven't touched any Arduino development since then.

Hopefully people will still have uses for this project.

MrPointer, 2017-12-14:
@queezythegreat Oh! Your'e alive! 😆 
Great to here that from you, Thomas. We have a lot of respect for your amazing work and contribution which led us to where we are today! Long live the **Arduino-CMake** community 😃